Jalle19/eda-modbus-bridge #10: Family type reported incorrectly

tomrosenback, 2022-01-08:
I have got a Pandion unit, but is here reported as a Pingvin. 

Have not yet debugged it but will do in the coming days.

Jalle19, 2022-01-08:
Interesting, that would suggest the Modbus register that indicates the family is always set to 0. See if you can find another register that holds useful information about the unit.

If you find a Modbus registry map for Pandion units, please consider adding it to the docs/ directory.

tomrosenback, 2022-01-09:
Noticed on the display unit that it displayed Pingvin there aswell, even if the unit is a Pandion, so this addon worked correctly.

The reason for this might be that I had the motherboard replaced in the unit quite some time ago and maybe then this one was programmed as a Pingvin.

I now wrote 1 to the family type register and now then Pandion came up in the summary view aswell.

closed 2022-01-09 by tomrosenback

Jalle19, 2022-01-09:
Haha, nice! I didn't know the register was writable.

tomrosenback, 2022-01-09:
Yup, all the registers / adresses with W in the Write column is writeable. :D