Jalle19/eda-modbus-bridge #17: Make HTTP service optional

tomrosenback, 2022-01-11:
In Home Assistant HTTP is not needed as everything is handled through MQTT. Make the parameter optional and disable HTTP endpoints if not provided.

Jalle19, 2022-01-11:
Good idea

Jalle19, 2022-01-11:
I personally use the HTTP in combination with https://github.com/Jalle19/eda-influxdb to store the readings in InfluxDB. Now that there's MQTT support this could be done differently, but I guess the work is already done so no point in changing it anymore.

tomrosenback, 2022-01-11:
Yes I figured you did so. Oh that makes me realize that I removed the HTTP ports from the config.json, which will as it stands break any HTTP calls. Can add that back.

Jalle19, 2022-01-11:
> Can add that back.

Please do :D

tomrosenback, 2022-01-11:
Did already :D

PR #20 (2022-01-11): Make MQTT discovery optional, make HTTP optional, update README MQTT section
Closes #13, closes #15, closes #17
commit 086d0e3f (2022-01-11): Make the HTTP server optional 

Closes #17
closed 2022-01-11 by Jalle19