Jalle19/eda-modbus-bridge #19: Expose device information over MQTT as well

Jalle19, 2022-01-11:
This would bring feature parity to the MQTT and HTTP APIs

tomrosenback, 2022-01-11:
Which specific device information would you like to see in? And where?

Jalle19, 2022-01-11:
These values:

```
eda/deviceInformation/fanType "EC"
eda/deviceInformation/coolingTypeInstalled null
eda/deviceInformation/heatingTypeInstalled "EDE"
eda/deviceInformation/familyType "Pingvin"
eda/deviceInformation/serialNumber 0
eda/deviceInformation/softwareVersion 2.17
```

Jalle19, 2022-01-11:
Maybe these should be published _once_ with retain set? I'm not that familiar with MQTT yet so I'm not sure if it's desirable here.

tomrosenback, 2022-01-11:
Maybe, question is what they are needed for as the information is shown in the Device info box and is already part of the device information in the discovery?

commit 4af96172 (2022-01-11): Publish device information to MQTT too

Closes #19
closed 2022-01-11 by Jalle19