Jalle19/eda-modbus-bridge #36: Try to reconnect to MQTT broker if disconnected

Jalle19, 2022-01-21:
Woke up and the MQTT part didn't work at all. Unfortunately I couldn't get at the logs since I have HTTP request spamming the log, and Home Assistant doesn't seem to store (or at least expose) logs that far back.

tomrosenback, 2022-01-21:
This should be handled by the MQTT library, check https://github.com/mqttjs/MQTT.js#about-reconnection

Jalle19, 2022-01-22:
It sounds like that section is only relevant when using the library from a browser context (in which case WebSockets are involved). I'll have to do some testing.

tomrosenback, 2022-01-24:
`mqttClient.on('connect', async () => {
    console.log("MQTT broker connected")
})

mqttClient.on('close', async () => {
    console.log('MQTT broker closed connection')
    
    setTimeout(async () => {
        console.log('Reconnect to MQTT broker')
        client.reconnect()
    }, 5000)
})`

Made some quick and dirty tests locally, above at least works when the broker dissappears after the first successful `connectAsync`. Does not handle the failed connection in the catch part.

Jalle19, 2022-01-24:
That probably on tries to reconnect once?

Jalle19, 2022-01-24:
https://github.com/mqttjs/MQTT.js/pull/101#issuecomment-49051189 maybe this is the actual issue?

tomrosenback, 2022-01-24:
> That probably on tries to reconnect once?

Could you please elaborate? mqttClient.on('close'... happens everytime a reconnect fail.

tomrosenback, 2022-01-24:
> [mqttjs/MQTT.js#101 (comment)](https://github.com/mqttjs/MQTT.js/pull/101#issuecomment-49051189) maybe this is the actual issue?

You are refering to the comment about the fact that after the client has reconnected it don´t subscribe to the topics previously subscribed? I thought you had problems also with the publishing?

Also it seems the MQTT really does default try to reconnect every second.

Jalle19, 2022-01-24:
Yeah, maybe I had some other issue, it's hard to tell since I never could get the complete log :/

tomrosenback, 2022-01-25:
Noticed after updating to 1.1.0 of the HA addon https://github.com/Jalle19/home-assistant-addon-repository which uses 2.1.0 of eda-modbus-bridge the MQTT connection failed on startup of the addon causing MQTT connection remaining disconnected until manual restart of addon. Restarting the addon manually helped in this case.

commit 367bd5ea (2022-02-08): Log MQTT reconnect attempts

Closes #36, at least for now
closed 2022-02-08 by Jalle19