Jalle19/eda-modbus-bridge #38: issue with nodejs 12.22.5

Sebastian930, 2022-01-22:
I have Raspberry Pi and latest Raspbian OS installed. Nodejs is installed from Raspbian package repository.

```
root@rasp02:~# cat /etc/debian_version
11.2
root@rasp02:~# nodejs --version
v12.22.5
```

eda-modbus-bridge (tested with latest and with 2.0.0 release) gives syntax error on start:

```
root@rasp02:~/eda-modbus-bridge# nodejs eda-modbus-bridge.mjs
file:///root/eda-modbus-bridge/app/modbus.mjs:314
        ][familyTypeInt] ?? 'unknown'
                          ^

SyntaxError: Unexpected token '?'
    at Loader.moduleStrategy (internal/modules/esm/translators.js:133:18)
```

I don't know shit about nodejs, but after removing "?? 'unknown'" from return statements, the program starts and seems to be working ok. And yes, I read that the program has only been tested with nodejs 14.x, but I thought compatibility with Raspbian might be useful for some users.

Thanks for the great software! :) My ventilation unit is Pandion (year 2012) and I can confirm that it works fine with eda-modbus-bridge.

tomrosenback, 2022-01-22:
I would recommend updating nodejs instead of removing the ??

Try `sudo apt-get update` followed by `sudo apt-get install nodejs` to install most recent version available for raspbian.

Sebastian930, 2022-01-22:
Hi Tom,
v12.22.5 is the latest nodejs for Raspbian (and for Debian too)
https://packages.debian.org/stable/nodejs

Jalle19, 2022-01-22:
> but I thought compatibility with Raspbian might be useful for some users.

Yeah, this incompatibility is unintentional and it would definitely be nice if the software worked on Raspbian out of the box. I'll see what I can do.

> Thanks for the great software! :) My ventilation unit is Pandion (year 2012) and I can confirm that it works fine with eda-modbus-bridge.

Awesome! How'd you find the project?

Jalle19, 2022-01-22:
https://github.com/Jalle19/eda-modbus-bridge/runs/4908028748?check_suite_focus=true indeed it fails

tomrosenback, 2022-01-22:
> Hi Tom, v12.22.5 is the latest nodejs for Raspbian (and for Debian too) https://packages.debian.org/stable/nodejs

Oh, I just remembered I had installed it myself trough apt-get on debian, it seems I haven´t but instead I have used nodesource https://github.com/nodesource/distributions/blob/master/README.md

My fault.

Jalle19, 2022-01-22:
@Sebastian930 can you try the https://github.com/Jalle19/eda-modbus-bridge/tree/nodejs12 branch?

Sebastian930, 2022-01-22:
> 
> Awesome! How'd you find the project?

IIRC I googled Enervent EDE modbus things couple weeks ago, I found this project and saved link. Modbus USB adapter arrived yesterday and today I had some time to test the adapter.

> Oh, I just remembered I had installed it myself trough apt-get on debian, it seems I haven´t but instead I have used nodesource https://github.com/nodesource/distributions/blob/master/README.md
> 
> My fault.

No problem :). I was also aware that newer nodejs is available from "external repository", but as @Jalle19 wrote, out of the box compatibility with Raspbian would be nice.

> @Sebastian930 can you try the https://github.com/Jalle19/eda-modbus-bridge/tree/nodejs12 branch?

This one works out of the box on Raspbian 11. Thanks a lot! Super fast fix 👍

commit 5b0f05aa (2022-01-22): Update CHANGELOG and README

Fixes #38
closed 2022-01-22 by Jalle19