Jalle19/eda-modbus-bridge #42: Alarm names and descriptions

JohanElmis, 2022-02-09:
Hi,
In the AVAILABLE_ALARMS definition we start using inlet and outlet - which is is terms that are not used elsewhere (and not that commonly used when talking about FTX) - but more with ventilation that doesn't have the the Heat recovery (just in/out). 

In the English Modbus definitions - they have names that are more similar to the rest of the code: 
Outside -> Supply Air and 
Extract Air -> Exhaust Air,  even if I personally like "Waste Air" better than Extract, that some other manufacturers use.
Would it cause issues somewhere for example HomeAssistant if we changed them to the Enervent (To match the names in the English Modbus list)?

I can live with either - so this is more a question than an issue.
I can change them - if you think it's good and will not create problems.

tomrosenback, 2022-02-09:
I will cause the need to remove the integration and re-install it, nothing more. If someone have added automations that are dependant on the alarms it will make these not work. But, I do think that we should change the names to match what is in the english Modbus list, I did not have that in my hands when I tried to translate some cryptic shortened finnish :D Ok @Jalle19 ?

Jalle19, 2022-02-10:
Yeah, we didn't have an official English document to go on when we came up with the current names so they are what they are.

> It will cause the need to remove the integration and re-install it, nothing more

I believe the alarm device names will be updated as long as we don't change the "unique ID" that we use when we publish MQTT discovery information?

@JohanElmis feel free to make a PR!

tomrosenback, 2022-02-10:
If it is to be changed, I would prefer to change the unique ID aswell, just so that they follow same naming convention.

Jalle19, 2022-02-10:
Yeah, I guess. Either way is fine by me.

JohanElmis, 2022-02-11:
Ok, I will try to create that this afternoon.

Jalle19, 2022-02-13:
Closed by #44

closed 2022-02-13 by Jalle19