Jalle19/eda-modbus-bridge #43: Different publish intervals for different information

Jalle19, 2022-02-10:
E.g. the "device information" stuff never changes, so it's just a waste of resources to publish those every $publishInterval seconds.

JohanElmis, 2022-02-10:
Agree, not sure if we should split it up further so we don't read stuff that we don't need either.
I want to get some more values - for example where I can see if the Electrical heater is running (and at what power level).
If it gets too cold outside - and the temp is set to high it may start (or just temporarily to dry out something) - but I would rather set down the temp a degree than to run with electrical heater.
I think I have seen this info in a bit-field. but I will look into that again.

Jalle19, 2022-02-11:
The "LTO I-cascade" value shows you thay. 0 means the drum isn't spinning at all, 100 means it's nominal, anything between 100 and 200 means the heater is being used.

JohanElmis, 2022-02-11:
Ok, I didn't know that. I will investigate more.
The state-values shows a lot of nice info - but it's a bit-field - and only status info.
But I guess that we could break it out as individual status flags.
I will do a test on this - and then we can look at it. 
Could be nice to see this in HA as well I guess.

Jalle19, 2022-02-11:
I'm not sure which state values you're referring to now?

Jalle19, 2022-02-11:
If you figure out how to detect when the device is in "humidity boost" mode let me know, I haven't been able to figure it out. Same thing with centrql vacuum etc.

JohanElmis, 2022-02-12:
I think I can.. will play with this tomorrow.

Jalle19, 2022-02-14:
Nice, I've missed those registers when skimming through the documentation.

PR #51 (2022-03-12): Publish device information once only
Fixes #43 

@JohanElmis maybe this helps your case a little bit?

@tomrosenback feel free to review and/or test
closed 2022-03-12 by Jalle19