Jalle19/eda-modbus-bridge #45: Reading-label errors

JohanElmis, 2022-02-12:
Hi, I'm curious to know what you get as reading for supply, waste and exhaust temperatures.
The names are correct according to the Modbus guide.
But it doesn't make sense that the exhaust air temp is higher than waste (my house is heated..)
Does other have the same issue.
Maybe my old firmware have this messed up.
I will double check on the EDA control panel - or if the temp-sensors have been mixed at some point.

`readings: {
freshAirTemperature: 4.7,
supplyAirTemperatureAfterHeatRecovery: 19.7,
supplyAirTemperature: 19.8,
wasteAirTemperature: 8.1,
exhaustAirTemperature: 22.7,
exhaustAirTemperatureBeforeHeatRecovery: -40,
exhaustAirHumidity: 29,
heatRecoverySupplySide: 83,
heatRecoveryExhaustSide: 81,
heatRecoveryTemperatureDifferenceSupplySide: 15,
heatRecoveryTemperatureDifferenceExhaustSide: 14.6,
mean48HourExhaustHumidity: 28,
`

JohanElmis, 2022-02-12:
This is really getting more and more complicated the more I look at it.
https://en.wikipedia.org/wiki/Heat_recovery_ventilation
Warm air from the house is Extracted, and the heat is "collected" in the exchanger and used to heat the incoming air. This far it's all good.
The outgoing air - leaving the FTX to the outside - is called Exhaust (just like the gases on a ICE car - through the exhaust pipe).
I found several companies and articles the other day where they used the Waste Air for the air sucked out from the house (which would then be the same as Extract).. but I can't find anything supporting that today. No mentions of Waste air at all. (Maybe that term is only used for building that doesn't have ventilation with heat exchanger.)
I also found some really confusing graphs: https://images.app.goo.gl/bWBUrkzQUVNz8RCE6 :-)
This however says that it's the exhaust. http://www.ventilation-ftx.se/wp-content/uploads/2012/06/FTX-rotor.jpg
Seems like many are using their own names - or using existing names wrong.
If waste air is correct to use with FTX - and its really the air going out to the outside after the exchanger, then the alarm-messages that I renamed is wrong.

JohanElmis, 2022-02-12:
5: { name: 'TE30ExtractAirCold', description: 'TE30 Waste air cold' },
This will trigger if the temp goes under 15deg for 10 mins. (Which the exhaust-air does a lot.)

Jalle19, 2022-02-13:
Yeah, as I commented in the pull request that particular alarm should be:

```
5: { name: 'TE30ExhaustAirCold', description: 'TE30 Exhaust air cold' },
```

I don't think there's any alarm related to the waste air temperature.

Jalle19, 2022-02-14:
@JohanElmis is there still something unclear or wrong or can we close this?

Jalle19, 2022-03-04:
Closing this

closed 2022-03-04 by Jalle19