Jalle19/eda-modbus-bridge #48: Add a "master alarm" entity

Jalle19, 2022-03-03:
There are a ton of different alarms, so it can be difficult to see if any of them are active. We could add a master alarm that is on when any alarm is on, and off when all alarms are off.

@JohanElmis @tomrosenback what do you think about this?

tomrosenback, 2022-03-03:
A have grouped the alarms in HA, this give same effect as described here. I am fine with adding it here.

Jalle19, 2022-03-03:
@tomrosenback how did you group them?

tomrosenback, 2022-03-04:
Like this I have it configured with the 2.1.0 release in a separate groups.yaml file

`eda_alarms:
  name: EDA alarms
  icon: mdi:exclamation-thick
  entities:
    - binary_sensor.eda_coolingfault
    - binary_sensor.eda_emergencystop
    - binary_sensor.eda_firerisk
    - binary_sensor.eda_heaterfault
    - binary_sensor.eda_heaterpressureswitch
    - binary_sensor.eda_ltofault
    - binary_sensor.eda_hpfault
    - binary_sensor.eda_inletfanpressureabnomaly
    - binary_sensor.eda_inletfilterdirty
    - binary_sensor.eda_outletfanpressureabnomaly
    - binary_sensor.eda_outletfilterdirty
    - binary_sensor.eda_returnwatercold
    - binary_sensor.eda_servicereminder
    - binary_sensor.eda_te10inletafterheatercold
    - binary_sensor.eda_te10inletafterheaterhot
    - binary_sensor.eda_te20roomhot
    - binary_sensor.eda_te30outletcold
    - binary_sensor.eda_te30outlethot
    - binary_sensor.eda_te5inletafterheatexchangercold

`

Jalle19, 2022-03-04:
Nice, didn't know entities could be grouped together like that. I'll try it out, sounds better than hacking a "master alarm" into this software.

closed 2022-03-05 by Jalle19