Jalle19/eda-modbus-bridge #56: Repeated alerts after enabling this bridge

odelma, 2022-10-10:
Hi, first of all thank you for this excellent bridge 👍 

The bridge seems to be working well, but I have a strange issue that started at the same time as I installed it. I've been getting the following alert approximately every few days (sometimes twice a day, sometimes once per week):

TE20 Yläraja
Palovaara
Tuloilma on kuumaa.

This stops the machine which is quite inconvenient and I am quite certain it is not a "true" alert (in reality the air temp is normal). Machine is from 2009, Enervent Pingvin Eco EDE and I have not seen this error before since starting to use the bridge. Any ideas what could be causing it and how to mitigate it?

Jalle19, 2022-10-10:
Unfortunately I have the same issue. I haven't found a workaround. Sometimes it can be many months between occurences so I've managed so far.

Jalle19, 2022-10-10:
We could try tweaking how often the alarm related registers are read to see if the issue goes away. Now that we have two confirmed cases of this happening with the same model I could also contact Enervent and ask for advice.

tomrosenback, 2022-10-10:
I have not experienced this on my Pandion, I have however had the interval set to 60s

Jalle19, 2022-10-10:
It feels lile this could be a Pingvin specific issue. @JohanElmis have you experienced this problem?

Jalle19, 2022-10-10:
@odelma can you try changing the publish interval setting to 30 seconds?

Jalle19, 2022-10-10:
I sent this to Enervent, let's see what they say:
> Hej!
> 
> Jag har gjort ett program för att styra Enervent ventilationsmaskiner från Home Assistant (en populär hemautomationsplattform) och har upptäckt ett problem som det visar sig att även andra har, nämligen:
>
>Då man läser alarmregistren på en Pingvin Eco EDE så kan maskinen få för sig att aktivera "TE20 Yläraja, palovaara" alarmet, vilket leder till att hela enheten stängs av. Det här kan hända flera gånger på dygn, eller så kan det gå flera månader mellan alarmen.
>
>https://github.com/Jalle19/eda-modbus-bridge/issues/56 här är en annan person som har exakt samma maskin och konfiguration (Pingvin Eco EDE) som jag och som upplever samma problem.
>
>Har ni någon erfarenhet av just den här typen av problem? Finns det nånting jag kan göra för att förhindra att enheten går i alarmläge? Hittills har jag läst av alla 20 "alarmslottar", 7 register i taget, var 10:e sekund. Kan det här påverka på något sätt?
>
>Ytterst tacksam för svar!

Med vänliga hälsningar
Sam Stenvall

odelma, 2022-10-10:
> @odelma can you try changing the publish interval setting to 30 seconds?

Sure thing, changed it to 30 seconds now. Thanks for the prompt reply. I'll report back after a week or so (or sooner if it keeps happening more frequently) if this did the trick. Also hoping Enervent will reply to your query 👍

Jalle19, 2022-10-11:
Got a reply from Enervent:
> Ted kan hända att TE20 alarm kommer när man har Modbus kopplat om man inte har en temperaturgivare eller ett motstånd kopplad till temperaturgivar-anslutningen på styrpanelen. Man kan använda en NTC10 givare eller ett motstånd på 10kohm för att förhindra detta fenomen.

Translation: the TE20 alarm can be tripped when Modbus is used and no temperature sensor or resistor is plugged in to the main board. Adding an NTC10 sensor or a 10 kiloohm resistor should alleviate the issue completely.

Guess I'll be buying a resistor then.

odelma, 2022-10-21:
> Sure thing, changed it to 30 seconds now. Thanks for the prompt reply. I'll report back after a week or so (or sooner if it keeps happening more frequently) if this did the trick. Also hoping [...]

So far so good, no more errors since then ☺️

odelma, 2022-10-24:
I think that at least from my viewpoint this can be closed. Enervent has provided (a sort of) explanation for the issue and from the looks of it, setting the interval to 30(+) seconds fixes at least the frequent alerts.

closed 2022-10-24 by odelma