Jalle19/eda-modbus-bridge #57: Compatibility with Pingvin 120 ECE

nhallfast, 2022-10-17:
Hi,

I have a Enervent Pingvin 120 with ECE control, not EDA (example of ECE control panel: https://www.enervent.fi/wp-content/uploads/2015/07/Exvent_ECE_paneeli_2011-743x576.jpg).

Do you know whether it's possible to use your bridge for controlling it?

I have a USB adapter (https://www.waveshare.com/usb-to-rs485.htm) and did some quick tests with the bridge, but they didn't seem too promising (GET to `/summary` failed.).

Thanks in advance!

Jalle19, 2022-10-17:
Let's start with a stack trace, of the unit has a "Freeway" port then it should be compatible, at least to some extent.

nhallfast, 2022-10-17:
Thanks for the speedy response.

The stacktrace I get is:
```
node:internal/process/promises:279
            triggerUncaughtException(err, true /* fromPromise */);
            ^

[UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "#<TransactionTimedOutError>".] {
  code: 'ERR_UNHANDLED_REJECTION'
}
```
I have tried different values for the slave-argument at launch, but they all produce the same stacktrace. 

I have a setup where the machine itself has a control panel (with the free port) to that I have connected another control panel, which in turn has another free port and the dongle is connected to this port. So I assume the slave value should be either 2 or 3 (as per https://web.archive.org/web/20201020102005/http://ala-paavola.fi/jaakko/doku.php?id=pingvin)

Could it be that I have messed up the connection of the cable to the USB dongle?
I have currently:
- Yellow to GND
- Red to A+
- Green to B-

Jalle19, 2022-10-17:
The unused port on the control panel is for daisy chaining control panels together - they cannot be used for Modbus communication. You need to connect your RJ10 cable directly to the maim board on the Pingvin unit, assuming it has a port labeled "Freeway".

Jalle19, 2022-10-18:
Judging by that schematic you can just as well use the daisy-chain port on your control panel - no need to attach your USB dongle directly to the mainboard. EDA units have a third port that's different to the OP1 and OP2 ports, but this one doesn't seem to have that.

You could try some generic Modbus register explorer tool on your laptop to see if the unit responds at all when reading any registers. I'm not entirely sure that the ECC units can be controlled via Modbus so it could be a dead end.

nhallfast, 2022-10-18:
Alright, will try that. Thanks very much for your help, greatly appreciated!
I'll close this ticket now, I can get back to you if I get this working. In case there are any other poor suckers out there with this control panel on their Pingvins.