Jalle19/eda-modbus-bridge #58: Is it possible to set Eco Mode on?

kailampi, 2022-11-14:
Hi,

I found this great integration few weeks ago and it works like a charm once I figured it out how the Freeway bus works.. I have automated the away / home -setting depending on my home alarm system state. 

I'd prefer to use Pingvin's Eco Mode as it adjusts supply air temperature to maintain efficient at 100. However, once my automation returns from away mode to normal mode, it always returns normal without eco. I have not found any setting to set Pingvin to eco mode via modbus. Do you know is that even possible?

Jalle19, 2022-11-14:
I don't even know what eco mode is 🤔 What does it look like on the control panel?

kailampi, 2022-11-14:
It is something you can set on Enervent eAir control panel: [https://www.enervent.fi/kotilampo-ohjaus/](url)

See the last picture on below. There is a leaf icon top right corner. In the picture it is off, because no green light. When on, the leaf icon is green. I can set it via panel, but have not find any ways with this bridge. I have not (yet) looked does it exists in Enervent modbus registers.

kailampi, 2022-11-14:
I don't know how to use this, but seems like there is modbus coil register for the eco mode:

coil 40 | COIL_ECO_MODE |   | 1 |   | Eco mode | Eco mode desc | Eco mode | Eco mode desc.
-- | -- | -- | -- | -- | -- | -- | -- | --

Jalle19, 2022-11-15:
Interesting, it seems you have a different control panel (eWind?) and it uses some registers I haven't encountered before.

Can you paste the output of `/summary` from your installation? E.g. if your bridge is running on 192.168.1.100 on port 8080 the URL would be `http://192.168.1.100:8080/summary`. That would help me get a better understanding of what works and what doesn't work when it comes to eWind units. I'm especially interested in the `deviceInformation` object.

kailampi, 2022-11-15:
Here is the output:

{"flags":{"away":false,"longAway":false,"overPressure":false,"cookerHood":false,"centralVacuumCleaner":false,"maxHeating":false,"maxCooling":false,"manualBoost":true,"summerNightCooling":false},"readings":{"freshAirTemperature":2.9,"supplyAirTemperatureAfterHeatRecovery":17.8,"supplyAirTemperature":17.2,"wasteAirTemperature":7,"exhaustAirTemperature":18.8,"exhaustAirTemperatureBeforeHeatRecovery":-36.2,"exhaustAirHumidity":50,"heatRecoverySupplySide":93,"heatRecoveryExhaustSide":74,"heatRecoveryTemperatureDifferenceSupplySide":0,"heatRecoveryTemperatureDifferenceExhaustSide":0,"mean48HourExhaustHumidity":47,"cascadeSp":400,"cascadeP":180,"cascadeI":100,"overPressureTimeLeft":10,"ventilationLevelActual":90,"ventilationLevelTarget":30},"settings":{"overPressureDelay":10,"awayVentilationLevel":20,"awayTemperatureReduction":2,"longAwayVentilationLevel":20,"longAwayTemperatureReduction":3,"temperatureTarget":18},"deviceInformation":{"fanType":"EC","unitType":"Family","coolingTypeInstalled":null,"heatingTypeInstalled":"EDE","familyType":"Pingvin","modelType":"Pingvin","serialNumber":4,"softwareVersion":1.18,"modelName":"Pingvin eco EDE"},"alarmHistory":[{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-10-09T14:49:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-10-09T16:04:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-10-09T12:56:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-08-22T18:11:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-08-16T18:30:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-12-15T10:18:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2022-02-21T13:00:00.000Z"},{"name":"TE20RoomTempHot","description":"TE20 Room temperature hot","state":0,"date":"2021-12-06T16:27:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-10-19T17:06:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-08-18T18:38:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-07-24T05:18:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-03-22T20:20:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-03-06T21:11:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-02-17T22:28:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-02-09T20:02:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2021-02-06T23:08:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2020-12-18T14:27:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2020-08-10T16:04:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2020-11-30T17:31:00.000Z"},{"name":"ServiceReminder","description":"Service reminder","state":0,"date":"2020-11-02T14:11:00.000Z"}],"deviceState":{"normal":false,"maxCooling":false,"maxHeating":false,"emergencyStop":false,"stop":false,"away":false,"longAway":false,"temperatureBoost":false,"co2Boost":false,"humidityBoost":false,"manualBoost":true,"overPressure":false,"cookerHood":false,"centralVacuumCleaner":false,"heaterCooldown":false,"summerNightCooling":false,"defrosting":false}}

Jalle19, 2022-11-15:
Interesting, I have the same unit but a different firmware version and control panel. I can enable eco mode but it's not reflected on the control panel. I'll try enabling it over night and will check my graphs in the morning to see if the temperature setpoint has changed during the night.

kailampi, 2022-11-15:
Actually my device is Pingvin MDE, not EDE. Fortunately the Freeway bus seems to be quite similar.

Jalle19, 2022-11-15:
I can't find "Pingvin MDE" anywhere, are you sure that's the right model number? It should say on a sticker on the machine itself.

kailampi, 2022-11-15:
Yes, the sticker states exactly Pingvin MDE. In some reason it has been sold according to the control panel's name, like Pingvin eAir. This is my device: https://www.enervent.fi/product/pingvin/

Jalle19, 2022-11-15:
I think I figured it out. I'll update the logic that generates the model name.

@JohanElmis @tomrosenback what does the software version field return on your devices? I'm trying to determine whether 1.x == MD automation and 2.x == EDA.

tomrosenback, 2022-11-16:
Mine is a Pandion EDA, which returns 5.54 as software version.

Jalle19, 2022-11-16:
Okay. I'm wondering how to detect which type of automation is used :thinking:

Jalle19, 2022-11-16:
@tomrosenback what's the value of "serial number"? The old documentation says that the registers holds a serial number, but newer documentation says it's the hardware revision.

tomrosenback, 2022-11-16:
`"deviceInformation": {
    "fanType": "EC",
    "unitType": "Family",
    "coolingTypeInstalled": null,
    "heatingTypeInstalled": "EDE",
    "familyType": "Pandion",
    "modelType": "Pandion",
    "serialNumber": 0,
    "softwareVersion": 5.54,
    "modelName": "Pandion eco EDE"
  },`

Jalle19, 2022-11-16:
Could be that 0 == EDA, but it's hard to tell without access to more units. I think we'll let it be for now and assume we can't distinguish between these.

JohanElmis, 2022-11-16:
I have a 12 year old LTR-6 eco EDE, serialNr: 32313, software version 1.97
Not actively using this library yet. -but I want to switch over to this. Just need some time.
I have been pulling metrics from the FTX for many years. using another hack I did, so if I don't want gaps, I need to still get the data and send it to my storage. (Which is not hard as they are in Mqtt, so I just need to get them and remap the metric names.

Jalle19, 2022-11-16:
Thanks for the report!

Jalle19, 2022-11-16:
Support has been merged to master. I'll make a release soon.

closed 2022-11-16 by Jalle19

Jalle19, 2022-11-18:
Tried eco mode on my old Pingvin and it doesn't really seem to do anything. I guess it needs newer firmware.

kailampi, 2025-12-17:
Hi, I don't know where to find that deviceState, but from my HomeAssistant I see this, while eco is on. All numerous other states are false.

[image]

Jalle19, 2025-12-18:
I'll change the "eco" switch to be a normal settings switch and not a "mode" switch. That means it can be independently toggled regardless of the mode the unit is in. Will make it easier to implement #150 .