Jalle19/eda-modbus-bridge #66: Adapt to recent entity naming changes in Home Assistant

Jalle19, 2023-08-05:
https://www.home-assistant.io/blog/2023/08/02/release-20238/#breaking-changes

https://developers.home-assistant.io/docs/core/entity/#has_entity_name-true-mandatory-for-new-integrations

Jalle19, 2023-08-06:
Not really sure what there is to do here. It seems like the entity name and its "friendly name" are now separate, and Home Assistant auto-generates the friendly name as "deviceName + entityName", e.g. `Enervent Pingvin eco EDE/MDE Normal`. This looks clumsy but I'm not sure there's anything we can do better.

@tomrosenback do you have any suggestions?

tomrosenback, 2023-08-06:
I am using Zigbee2Mqtt addon aswell and for entities /devices coming from there I got a warning about the breaking change. I did not however see any entity from enervent. Did you?

Here is another blog about how to adress the change.
https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

Jalle19, 2023-08-06:
No, I haven't gotten any warnings, the only thing that has changed is the name shown in dashboard cards for entities. On the device page everything looks the same as before.

tomrosenback, 2023-08-07:
I don´t think there is much to do here, as you pointed out it is clumsy, but on the other hand it seems like this is the way it was intended. I didn´t realize this change affected the entities of the Enervent device before I tried to add it to my dash, I have renamed anyways most of my entities on the dashboard so for me this issue is ok to close as won´t do.

Jalle19, 2023-08-07:
Yeah, I had renamed some but not all so now my dashboard looks awful :shrug: I guess this is intended behavior.

closed 2023-08-07 by Jalle19