Jalle19/eda-modbus-bridge #77: Installation fails with npm ERR! code 1

ctekn002xamk, 2023-08-10:
Device: Raspberry Pi with Raspberry Os Lite 64-bit

Logs:

npm ERR! code 1
npm ERR! path /opt/eda-modbus-bridge/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! make: Entering directory '/opt/eda-modbus-bridge/node_modules/@serialport/bindings/build'
npm ERR! CXX(target) Release/obj.target/bindings/src/serialport.o
npm ERR! make: Leaving directory '/opt/eda-modbus-bridge/node_modules/@serialport/bindings/build'
npm ERR! node:internal/modules/cjs/loader:1080
npm ERR! throw err;
npm ERR! ^
npm ERR!
npm ERR! Error: Cannot find module './tracker-group.js'
npm ERR! Require stack:
npm ERR! - /opt/eda-modbus-bridge/node_modules/are-we-there-yet/index.js
npm ERR! - /opt/eda-modbus-bridge/node_modules/npmlog/log.js
npm ERR! - /opt/eda-modbus-bridge/node_modules/prebuild-install/log.js
npm ERR! - /opt/eda-modbus-bridge/node_modules/prebuild-install/bin.js
npm ERR! at Module._resolveFilename (node:internal/modules/cjs/loader:1077:15)
npm ERR! at Module._load (node:internal/modules/cjs/loader:922:27)
npm ERR! at Module.require (node:internal/modules/cjs/loader:1143:19)
npm ERR! at require (node:internal/modules/cjs/helpers:110:18)
npm ERR! at Object.<anonymous> (/opt/eda-modbus-bridge/node_modules/are-we-there-yet/index.js:2:24)
npm ERR! at Module._compile (node:internal/modules/cjs/loader:1256:14)
npm ERR! at Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
npm ERR! at Module.load (node:internal/modules/cjs/loader:1119:32)
npm ERR! at Module._load (node:internal/modules/cjs/loader:960:12)
npm ERR! at Module.require (node:internal/modules/cjs/loader:1143:19) {
npm ERR! code: 'MODULE_NOT_FOUND',
npm ERR! requireStack: [
npm ERR! '/opt/eda-modbus-bridge/node_modules/are-we-there-yet/index.js',
npm ERR! '/opt/eda-modbus-bridge/node_modules/npmlog/log.js',
npm ERR! '/opt/eda-modbus-bridge/node_modules/prebuild-install/log.js',
npm ERR! '/opt/eda-modbus-bridge/node_modules/prebuild-install/bin.js'
npm ERR! ]
npm ERR! }
npm ERR!
npm ERR! Node.js v18.17.0
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.1
npm ERR! gyp info using node@18.17.0 | linux | arm64
npm ERR! gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args 'binding.gyp',
npm ERR! gyp info spawn args '-f',
npm ERR! gyp info spawn args 'make',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/opt/eda-modbus-bridge/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/root/.cache/node-gyp/18.17.0/include/node/common.gypi',
npm ERR! gyp info spawn args '-Dlibrary=shared_library',
npm ERR! gyp info spawn args '-Dvisibility=default',
npm ERR! gyp info spawn args '-Dnode_root_dir=/root/.cache/node-gyp/18.17.0',
npm ERR! gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args '-Dnode_lib_file=/root/.cache/node-gyp/18.17.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args '-Dmodule_root_dir=/opt/eda-modbus-bridge/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args '-Dnode_engine=v8',
npm ERR! gyp info spawn args '--depth=.',
npm ERR! gyp info spawn args '--no-parallel',
npm ERR! gyp info spawn args '--generator-output',
npm ERR! gyp info spawn args 'build',
npm ERR! gyp info spawn args '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/serialport.cpp:1:
npm ERR! ../src/./serialport.h:6:10: fatal error: nan.h: No such file or directory
npm ERR! 6 | #include <nan.h>
npm ERR! | ^~~~~~~
npm ERR! compilation terminated.
npm ERR! make: *** [bindings.target.mk:114: Release/obj.target/bindings/src/serialport.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:203:23)
npm ERR! gyp ERR! stack at ChildProcess.emit (node:events:514:28)
npm ERR! gyp ERR! stack at ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 6.1.21-v8+
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /opt/eda-modbus-bridge/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v18.17.0
npm ERR! gyp ERR! node-gyp -v v9.3.1
npm ERR! gyp ERR! not ok

Jalle19, 2023-08-10:
I was able to reproduce with an ARM64 AWS instance:

```
$ npm i
npm notice 
npm notice New minor version of npm available! 9.6.7 -> 9.8.1
npm notice Changelog: https://github.com/npm/cli/releases/tag/v9.8.1
npm notice Run npm install -g npm@9.8.1 to update!
npm notice 
npm ERR! code 1
npm ERR! path /home/ubuntu/eda-modbus-bridge/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=18.17.1 runtime=node arch=arm64 libc= platform=linux)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.1
npm ERR! gyp info using node@18.17.1 | linux | arm64
npm ERR! gyp info find Python using Python version 3.10.6 found at "/usr/bin/python3"
npm ERR! gyp http GET https://nodejs.org/download/release/v18.17.1/node-v18.17.1-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v18.17.1/node-v18.17.1-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v18.17.1/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v18.17.1/SHASUMS256.txt
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/ubuntu/eda-modbus-bridge/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/ubuntu/.cache/node-gyp/18.17.1/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/home/ubuntu/.cache/node-gyp/18.17.1',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/home/ubuntu/.cache/node-gyp/18.17.1/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/home/ubuntu/eda-modbus-bridge/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: not found: make
npm ERR! gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:10:17)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:57:18
npm ERR! gyp ERR! stack     at new Promise (<anonymous>)
npm ERR! gyp ERR! stack     at step (/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:54:21)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:71:22
npm ERR! gyp ERR! stack     at new Promise (<anonymous>)
npm ERR! gyp ERR! stack     at subStep (/usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:69:33)
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/node-gyp/node_modules/which/which.js:80:22
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
npm ERR! gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
npm ERR! gyp ERR! System Linux 5.19.0-1025-aws
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /home/ubuntu/eda-modbus-bridge/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v18.17.1
npm ERR! gyp ERR! node-gyp -v v9.3.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: /home/ubuntu/.npm/_logs/2023-08-10T05_53_02_613Z-debug-0.log
```

The problem is we're depending on a too old version of `serialport` which doesn't ship pre-built packages for ARM64 + Node.js 18.x.

Jalle19, 2023-08-10:
@ctekn002xamk try pulling the latest code (`git pull --rebase`) and then try again, should work now!

PR #78 (2023-08-10): Update modbus-serial to 8.0.11, thus updating serialport to 10.x
Should hopefully bring serialport to a version that has native builds available so node-gyp isn't required. Fixes #77 

> For Node.js version 12.0, the last functioning release was serialport@10.

After this we'll have to drop support for Node.js 12.x.
closed 2023-08-10 by Jalle19