Jalle19/eda-modbus-bridge #86: Modbus TCP/IP connection possible?

juntta, 2023-09-04:
Thanks for great project! :)

I have Enervent Pelican HP eAir and its connected to my LAN with ethernet cable. Unit has software version 1.29 and in Modbus settings menu it has option Modbus TCP. After enabling this I can access registers and coils with Modbus client! 

Is it possible to use Modbus TCP connection already with this app or is code changes needed?

Jalle19, 2023-09-04:
Currently only Modbus RTU is supported, but if you're willing to experiment I can probably make a test version that uses Modbus TCP instead.

juntta, 2023-09-04:
Sure, happy to help! 

Just let me know when to start testing!

Jalle19, 2023-09-06:
Can you try this branch: https://github.com/Jalle19/eda-modbus-bridge/tree/modbus-tcp ? Specify the device like this: `-d tcp://192.168.1.40:502`, assuming 192.168.1.40 is your ventilation unit and 502 is the port. Do let me know what the default port is so I can use a better example if necessary.

juntta, 2023-09-06:
After short testing TCP connection seems to work just fine. I connected to ventilation unit and used MQTT broker to publish sensor data. Subscribed to mqtt broker and received sensor data just fine.

Might be some extra sensor (f.ex. room temp sensor) data that wasnt there but have to test it further and submit different issue about those.

Anyway, thanks for this enhancement!

Jalle19, 2023-09-06:
That's good news! I'll make a pull request with initial official support for Modbus TCP then.

I don't personally have any optional sensors (room temperature, CO2 etc.) which is why those sensors are not supported at the moment, but I'll gladly add them if you're able to test that they work. These should ideally be optional, i.e. the sensors aren't published to Home Assistant if they're not present (reporting 0).

PR #87 (2023-09-07): Add Modbus TCP support
Closes #86
closed 2023-09-07 by Jalle19