Jalle19/eda-modbus-bridge #88: missing sensor data with Pelican HP eAir?

juntta, 2023-09-10:
It seems there are some sensors data missing (at least from mqtt side) with Pelican HP eAir modbus.

List of findings (source: eAirMD-modbus-register-list-public):
- hreg 46 average temperature from room sensors and control panel (TE20 room temperature...)
- hreg 17-28 Analog input voltage values & their calculated results 

First row is kind of self explanatiory. :)
Second: Analog input values are extra sensors (Co2, Humidity and room temperatures)

Jalle19, 2023-09-10:
Do you have any analog sensors/inputs? It will be hard to implement them correctly without someone who can verify the values.

The rest of the readings should be straight-forward to implement. I'm thinking the fan speed values could be exposed to Home Assistant as disabled by default sensors, since one is not supposed to change them nilly willy.

juntta, 2023-09-10:
I have 6 analog sensors: 2 co2's, 2 temperature and 2 humidity so glad to with this one too.

I think I saw somewhere in the registry spreadsheet that register values 1-50 are read-only values so disabled sensors sounds like smart move. As well reserved ones too since there was some cautionary text too about changing those too.

Jalle19, 2023-09-11:
It seems my machine was configured to have two RH sensors and two CO2 sensors, although none of those are actually present so the values reported have been 0 :shrug:

Jalle19, 2023-09-11:
Can you try the https://github.com/Jalle19/eda-modbus-bridge/tree/more-sensors branch? If you enable debug logging (`-v` or `--debug`) it should log something like this when it retrieves the readings:

```
2023-09-11T11:21:02.258Z [modbus] debug: AI sensor types:
{
  data: [ 0, 0, 0, 0, 0, 0 ],
  buffer: Buffer(12) [Uint8Array] [
    0, 0, 0, 0, 0,
    0, 0, 0, 0, 0,
    0, 0
  ]
}
2023-09-11T11:21:02.290Z [modbus] debug: Reading holding register address 23, length 6
2023-09-11T11:21:02.322Z [modbus] debug: Reading holding register address 511, length 7
2023-09-11T11:21:02.322Z [modbus] debug: AI sensor values:
{
  data: [ 0, 0, 0, 0, 0, 0 ],
  buffer: Buffer(12) [Uint8Array] [
    0, 0, 0, 0, 0,
    0, 0, 0, 0, 0,
    0, 0
  ]
}
2023-09-11T11:21:02.323Z [modbus] debug: AI sensor readings:
{}
```

In your case there should be something other than zeroes everywhere.

I also added `roomTemperatureAvg`, for me it reports 0 since I don't have any room temperature sensors (and despite what they say, OP1 doesn't seem to actually report any values).

juntta, 2023-09-12:
Thanks, I’ll try to test this tonight

Jalle19, 2023-09-12:
> Also fan speed (both supply and exhaust, hreg 3 & 4) seems to be missing

These registers correspond to OP3 and OP4 control panel temperature sensors and are 0 in my case, so I think it doesn't make sense to support these. This is the first time I've seen a discrepancy in the register lists for various devices.

juntta, 2023-09-12:
> These registers correspond to OP3 and OP4 control panel temperature sensors and are 0 in my case, so I think it doesn't make sense to support these. This is the first time I've seen a discrepancy in [...]

hmm, could it be so that there are different registers for different units (or product families/generations)? 

Fan levels are nice to have info so these can be ignored. But if there some important settings in some registers which refer to different place in registers depending about model, could there be different sources (registry listing) for different model? Sounds like a big change… :(

Jalle19, 2023-09-12:
Everything is possible but I would prefer to keep things simple.

juntta, 2023-09-12:
> 

Seems to work, mqtt explorer on macOs shows:
[image]

Jalle19, 2023-09-12:
Is humidity 2 supposed to be zero?

juntta, 2023-09-12:
Good point! Currently sensor is disconnected - second bathroom is under renovation :)

juntta, 2023-09-12:
> These registers correspond to OP3 and OP4 control panel temperature sensors and are 0 in my case, so I think it doesn't make sense to support these. This is the first time I've seen a discrepancy in [...]

I just compared this older registry file:
https://github.com/Jalle19/eda-modbus-bridge/blob/master/docs/EDA_Modbus_Registers_2011_09_14.pdf
with the newer models registry: 
https://github.com/Jalle19/eda-modbus-bridge/blob/master/docs/eAirMD-modbus-register-list-public.xlsx
and there definitely are some differences.

F. ex. those fan speeds are at registries 51 &. 52 in the older file and 3 & 4 in the newer. 50-103 are packed with data in the old one and newer one those registries are "reserved" (maybe empty). Lots of other similar differences as well so.

Anyway, most essential registries match with both so I'm happy with these. But if you want and have energy to develop this further, I can help to dig out data and test!

Thank you so far for your great work!

Jalle19, 2023-09-12:
I'll tidy up the support for analog inputs before making a pull request, thanks for testing! I'll have to add plumbing for Home Assistant too so the sensors show up there.

juntta, 2023-09-15:
I tested this once more and no issues raised!

closed 2023-09-16 by Jalle19