Jalle19/eda-modbus-bridge #96: Pingvin MDE eAir spesific register values

Ta802, 2023-12-06:
Could there be a possibility to add different modbus registers to be shown through addon?

Pingvin MDE eAir has holding register "1" (Room temperature sensor TE20) and "4" (Current exhaust fan speed) which I would be interested to get data from. Is there a way to make this possible while not breaking addon for others using different models.

eAir modbus register list can be found here: https://doc.enervent.com/out/out.ViewFolder.php?folderid=16&showtree=1

Jalle19, 2023-12-06:
Moved the issue to the correct repository

Jalle19, 2024-01-18:
@juntta can you test this branch and see if you get any values for supply and exhaust fan speeds (holding registers 3 and 4)?

juntta, 2024-01-20:
> @juntta can you test this branch and see if you get any values for supply and exhaust fan speeds (holding registers 3 and 4)?

Ok, I'll try to test it out this weekend!

Jalle19, 2024-01-20:
https://github.com/Jalle19/eda-modbus-bridge/pull/97 here, `more-sensors`

juntta, 2024-01-27:
Sorry, got sick last weekend:

[image]

Jalle19, 2024-01-27:
What firmware veesion do you have?

juntta, 2024-01-28:
Software version from the integration shows 1.42. Is that what you meant? The eAir software version is 2.14.

commit a6169780 (2024-02-03): Add optional sensors for control panel temperature and supply/exhaust

fan speeds

These values are not available on all units so they're limited to MD
units for now

Closes #96, replaces #97
commit 0a2a9bb8 (2024-02-03): Add optional sensors for control panel temperature and supply/exhaust

fan speeds

These values are not available on all units so they're limited to MD
units for now

Closes #96, replaces #97
commit 7839cdfe (2024-02-03): Add optional sensors for control panel temperature and supply/exhaust

fan speeds

These values are not available on all units so they're limited to MD
units for now

Closes #96, replaces #97
commit f4fa24ac (2024-02-07): Add optional sensors for control panel temperature and supply/exhaust

fan speeds

These values are not available on all units so they're limited to MD
units for now

Closes #96, replaces #97
closed 2024-02-07 by Jalle19