Jalle19/eda-modbus-bridge #100: Illegal data address 72

mhog79, 2024-01-15:
I have a Pandion ECO-EDE with EDA automation and the Enervent Freeway wifi module that also makes modes TCP possible. When I try to start the add-on I get the following error and the add-on will not start.

2024-01-14T19:35:33.245Z [modbus] error: Failed to read coil address 72, length 1
2024-01-14T19:35:33.252Z [main] error: An exception occurred: Error: Modbus exception 2: Illegal data address (register not supported by device)

Jalle19, 2024-01-15:
Are you connecting directly to the ventilation unit, or are you connecting to the Freeway unit via Modbus TCP?

mhog79, 2024-01-15:
I am connecting to the Freeway unit via Modbus TCP.

Jalle19, 2024-01-17:
Should be fixed in https://github.com/Jalle19/eda-modbus-bridge/tree/2.7.1, I'll post a comment here again when the addon has been updated too

closed 2024-01-17 by Jalle19

Jalle19, 2024-01-17:
Addon updated to 1.6.1 which contains this fix, please try it and report back whether it fixed the issue or not. There might be some other unsupported registers in there still.

mhog79, 2024-01-17:
That did it. Now it loads and I am able to see it in Home Assistant. Thank you very much!

Jalle19, 2024-01-18:
Very good, thanks for reporting back!