Jalle19/eda-modbus-bridge #109: Add Defrosting switch from coil address 1x0055?

c-mholm, 2024-02-12:
Hello, and thank you for this awesome bridge for the Enervent.

I use a Waveshare RS485 to Ethernet serial server talking to my Enervent Pingvin eco EDE.[image]
The serial number is zero probably because  I had to replace the logic board some yers ago due to moisture damage.

I wonder if it is possible to add a HA switch to toggle the defrosting function on/off? Now I use HA:s modbus integration to switch defrosting on and off, and I think that is making the Eda-modbus-bridge communication erratic sometimes. I'm using defrosting function to avoid freezing failure of the heat exchanger, I have had some at sub -20. At the same time I must be able to switch defrosting off when I put fire in the fireplace, or my house fills with smoke when the defrosting turns on and under pressure is built up in the house.

As a side note I have tried to get the Overpressure to work by just writing 1 to coil register 1x0003 but it doesn't work. If you have any hints for getting this to work I would be happy :).  I tried to look at your code but couldn't find anything useful with my lack of knowledge. I suspect the use of two separate instances that reads the Pingvins Modbus make the Eda-modbus-bridge unstable. I get errors in the log and a restart or two of the add-on resolves the issue for some time.

/Carl-Mikael Holm

Jalle19, 2024-02-12:
I'm not sure if turning defrosting off will fix your issue, in my case setting it to "on" just allows the unit to enter defrosting automatically when it feels like it.

Enabling overpressure mode using coil 3 should work. Did you try enabling it directly via Modbus from Home Assistant or did you use this software?

Jalle19, 2024-02-12:
Anyway, I can definitely add a switch for enabling/disabling LTO sulatus, if you'd find it useful.

c-mholm, 2024-02-12:
I am manually turning off defrosting when I'm putting fire in the fireplace. Defrosting turns of the fresh air fan, causing under pressure in the house when the exhaust fan dries/melts the heat exchanger with warm indoor air. This happens every 2 hours, see attached graph. 
[image]

This is the code I have been trying to turn on overpressure with. 
[image]
I have been fiddling with this Modbus stuff for a week now. I have been trying to enable overpressure from HA Modbus integration. I have tried with both coil address 3 and 4 and nothing changes the state from Home to overpressure or any other state, which is strange. It's a little bit confusing when the first address starts with a zero, have to trial and error. 

Your software is working, I can get overpressure with the switch.

I got rid of the erratic behaviour by increasing HA Modbus polling interval from 5sec to 30sec.

If you enable the Defrost switch in your software I can ditch mine and just use yours, ha ha.

Jalle19, 2024-02-12:
[image]

I'm actually seeing the same behavior (defrosting occurring every two hours). In my case though it's not enough to disrupt the fireplace, but that surely varies from house to house. My neighbor had to use over-pressure mode every time he used the fireplace and it turns out there was some dirt in the intake duct or something...

> Your software is working, I can get overpressure with the switch.

Good!

> I got rid of the erratic behaviour by increasing HA Modbus polling interval from 5sec to 30sec.

Yeah, if two Modbus masters try to speak to the same slave simultaneously, something is bound to go wrong sooner or later.

> If you enable the Defrost switch (Coil 55) in your software I can ditch mine and just use yours, ha ha.

Great, I'll add it!

closed 2024-02-12 by Jalle19

c-mholm, 2024-02-12:
Thank you so much.

My fireplace doesn't have a separate air intake duct for fresh air. That's why there will be under pressure in my house, and the whole house is pretty air tight. Even when the cooker hood fan is on I have to overpressure or open a window, the hood is not connected to the Pingvin...yet.

PS. Did you know we are related ;)

Jalle19, 2024-02-12:
I just published and update to the Home Assistant add-on so click "check for updates" there and install the latest version, then the "Defrosting allowed" switch should appear.

> My fireplace doesn't have a separate air intake duct for fresh air. That's why there will be under pressure in my house, and the whole house is pretty air tight. Even when the cooker hood fan is on I have to overpressure or open a window, the hood is not connected to the Pingvin...yet.

I don't have a separate intake either. I "solved" the cooker hood by having a button next to the fireplace, and when I press it it activates over-pressure mode _and_ disables power to the cooker hood (and starts a timer which re-enables power after six hours).

> PS. Did you know we are related ;)

No I did not! How?

c-mholm, 2024-02-12:
Thanks, I just updated and added Defrosting allowed. Testing later tonight.

We are little cousins, ask your dad he knows. And if he doesn't then I have mistaken you for someone else, but Im pretty confident.

PR #110 (2024-02-12): Add setting switch for "defrosting allowed"
Closes #109
commit 7fca05c8 (2024-02-12): Add setting switch for "defrosting allowed"

Closes #109