Jalle19/eda-modbus-bridge #112: Holding registers 3x0051 and 3x0052

pkuronen22, 2024-04-02:
These are fan speed controls for in and out air fans. They work with my 2014 Enervent Pingvin EDE but are not included in this project and according to documentation these are not supported. For some reason they work with my model which is just a basic EDE with heating coils. 

Example commands with modbus-cli using cheap Waveshare TCP-RTU bridge:

- Put air input to minimum speed: modbus <ip address> 52=20 -s 1
- Put air input to maximum speed: modbus <ip address> 52=100 -s 1

Can you include these in your implementation? Thanks for your efforts for this!

Jalle19, 2024-04-02:
I haven't tested these myself, I've only tested that register 53 doesn't work.

Out of curiosity, what software/firmware version does your Pingvin have?

pkuronen22, 2024-04-08:
I tried reading 3x599 for firmware version but got something like "5" as answer. How to get the firmware version properly?

Request: Could you include 3x0051 and 3x0052 fresh / waste air adjustment of 20-100 to the integration? 

Using modbus home assistant integration seems to not know how to do it straight. Have no idea someone decided to define the registry interactions with such manner where you can only write temperatures or some fan buttons. Command line script modbus-cli integration would work better than home assistant modbus integration in this case.

Jalle19, 2024-04-08:
Register 599 should give you a three-digit number, which you then divide by 100 to get the firmware version. You can also check it from your device's control panel, if you have one with a display.

pkuronen22, 2024-04-09:
> I got 558 so is it 5.58 then.
> 
> `modbus 192.168.xx.xx 599 -s 1 Parsed 0 registers definitions from 1 files 599: 558 0x22e `
> 
> Edit: i am using waveshare rs485 to tcp rail mountable converter. Just to let you know.

I got supply and waste air working using automation call service modbus but if you can include them in the integration it would be most useful. Thanks :)

Jalle19, 2024-04-09:
I'll see what I can do, I've been reluctant exposing settings that can let the user screw up their ventilation completely (e.g. by accidentally configuring a constant over-pressure), but I can definitely see why these would be useful to have. It's a shame there's no register for controlling the fan speed easily like on the control panels.

pkuronen22, 2024-04-09:
> I'll see what I can do, I've been reluctant exposing settings that can let the user screw up their ventilation completely (e.g. by accidentally configuring a constant over-pressure), but I can [...]

Quite understandable. I will check up on here if you manage to do it. Meanwhile I will screw up my system with my own glue.

Jalle19, 2024-04-19:
I experimented with these registers a bit today. Changing the ventilation level from the control panel doesn't change the values of registers 51 and 52. Changing register 53 changes what the control panel _displays_, but not the actual ventilation level.

I'm reluctant to expose registers 51 and 52 since AFAICT they should only be adjusted when the unit is commissioned, not during normal operation. If you figure out a way to make changes to register 53 "stick", i.e. emulate whatever the control panel is doing, let me know.

closed 2024-04-19 by Jalle19

pkuronen22, 2024-04-20:
I misled you here unintentionally. Found it out myself too that 51 and 52 control the relative percentages of waste and fresh air while 53 controls the overall air volume. Would be more logical to have one register for "balance" only as the air volume literally seems to go down when you increase 51 or 52 while 53 is at 100.

I am considering implementing 51-52 with one slider and 53 will be the master slider. 

The 53 is working normally as expected using the Waveshare RTU-Ethernet din module. I made a helper number limited between 20-100 and adjusting that triggers an automation that calls modbus service holding reg 53 with the helper input number value.

pkuronen22, 2024-04-20:
Sorry for crawling the topic a bit here but I stuck a RS485 dongle to my 2004 Enervent AC Pingvin RS485 port. I am getting definately data from there but so far nothing that makes sense. If you have any leads on any of it I would be grateful of that. I have read in a taloLogger thread that some old models may be using modbus but timings are not same but using some kind of "stream" mode and detecting modbus frames from the stream of data might work.

Jalle19, 2024-04-20:
The symptoms you describe sounds like what happens when you try to connect via the OP1 and OP2 ports instead of the Freeway port. But the unit could definitely just be too old, the earlist register lists I've used as reference have been dated ~2009.

pkuronen22, 2025-03-14:
I actually made a working esphome integration to 2004 Enervent AC by reverse debugging control panel traffic, giving esp32 another address and using it with another port and it works very well. Can share if you have any use.

Meanwhile I am struggling to get my esp32 to EDA unit RS485 modbus. For some reason I could read temperatures etc but could not write those values mentioned earlier, got occasional CRC errors and now i cannot get anything from this freeway port. Just sometimes single by "FC" or "F0". Did you happen to run into that?

Jalle19, 2025-03-14:
> Can share if you have any use.
Sure, why not!

> got occasional CRC errors and now i cannot get anything from this freeway port. Just sometimes single by "FC" or "F0". Did you happen to run into that?

I think I saw some CRC errors when I initially embarked on this journey, I think it was due to using non-grounded RS485 stick.

pkuronen22, 2025-03-14:
Thanks for the tip. I am using the ground (and 5v power) from RS485 but should i have it grounded directly to the Enervent unit ground as well?

Jalle19, 2025-03-14:
Yes

pkuronen22, 2025-03-14:
Still very quiet and not even CRC errors. Getting occasional E0 once a couple of minutes from freeway port. But op panel port 2 was very noisy. Will try more.