Jalle19/eda-modbus-bridge #122: Misleading name of exhaustAirTemperatureBeforeHeatRecovery reading

ajulk7, 2024-11-23:
I've been testing eda-modbus-bridge with "LTR-6 eco EDW" model (EDA version is 2.14), which has water circulating radiator as after heater. Works nicely with LTR-6, except that return water temperature is not available in /summary. That's an interesting reading for EDW owners. 

But it seems register value 3x0011 which is shown in summary with name "exhaustAirTemperatureBeforeHeatRecovery" is actually same value as the 3x0012 returning water value. I tried this by modifying the code to show 3x0012 also. There is same value in both registers. Control panel also shows the same reading as "Paluuvesi" (returning water).

In the Enervent registry listing, it says "X8" for both. This is a bit weird design from Enervent.

[image]

Electric diagram in the EDA manual says that X8 sensor is return water temperature sensor. And that the X8 sensor is only available in EDW models. I wonder what's the 3x0011 reading value for ED/EDE?

[image]

I think it would make sense to change the reading name "exhaustAirTemperatureBeforeHeatRecovery" to "returnWaterTemperature", to avoid confusion.

ajulk7, 2024-11-24:
Did more reading at the Enervent docs site. In PELICAN/PEGASOS eco PRO greenair HP (HP_2012_2_en.pdf) there is difference use for X8, "After coil exhaust air sensor" (in finnish version it's "poistoilmapatterin jälkeinen anturi"). But this does not match the name in registry listing either. 

[image]

Maybe it could be conditional, if device is EDW then show 3x0012 "returnWaterTemperature", else show 3x0011 "exhaustAirTemperatureBeforeHeatRecovery". Or just include both readings.

Jalle19, 2024-11-24:
Interesting find!

I think the best option would like you said be to make the current exhaustAirTemperatureBeforeHeatRecovery sensor optional and only enabled on ED/EDE models, then introduce another optional sensor returnWaterTemperature for EDW models.

Jalle19, 2024-11-24:
In the eWind register list, 3x0012 is mentioned as:
> Holding reg 12 | HREG_T_WR | Return water | Return water temperature in supply air heater (sensor TE45)

commit 1420600c (2024-11-24): Make exhaustAirTemperatureBeforeHeatRecovery optional, add optional returnWaterTemperature sensor

Fixes #122
closed 2024-12-10 by Jalle19