Jalle19/eda-modbus-bridge #131: Add support for holding register 3x0136 (Temperature control type)

nimikko, 2025-01-27:
Add support for holding register 3x0136 so that temperature control type can be changed. For example during summer I normally use exhaust air and during cold winter periods change it to inflow air so that there is no need to heat the whole house with the ventilation unit but you can still use all possible heat recovery.

Holding register from the documentation
3x0136 Lämpötila säätötapa IV-koneen lämpötilasäätötapa, 1=Tulosäätö, 2=Poistosäätö, 3=Huonesäätö

Jalle19, 2025-01-27:
Interesting, I've seen that register listed but have never understood what it actually does. I'll try implementing support for it!

> during cold winter periods change it to inflow air so that there is no need to heat the whole house with the ventilation unit but you can still use all possible heat recovery.

I do something similar by disabling "heating allowed" (coil 1x0054), that makes it use just LTO no matter how cold it gets.

nimikko, 2025-01-28:
Perfect, it works just like it should. Thank you a lot.

Jalle19, 2025-01-29:
Thanks, I'll make a new release soon

Jalle19, 2025-01-29:
https://github.com/Jalle19/eda-modbus-bridge/releases/tag/2.9.1

closed 2025-01-29 by Jalle19