Jalle19/eda-modbus-bridge #144: Issue when installed into Debian 12

juntta, 2025-03-16:
Installed this bridge to new Debian 12 virtual server since my old mac mini died and now running into problems. I wonder if anyone can help me out.

I'm getting these errors when running bridge from command line:

`/opt/eda-modbus-bridge/dist/eda-modbus-bridge.js: line 1: use strict: command not found
/opt/eda-modbus-bridge/dist/eda-modbus-bridge.js: line 2: syntax error near unexpected token `('
/opt/eda-modbus-bridge/dist/eda-modbus-bridge.js: line 2: `var __importDefault = (this && this.__importDefault) || function (mod) {'`

Tried to find quick fix and found some suggestions to add shebang into first row:
`#!/usr/bin/env node`

Then bumped into next issue with modbus.js file:
`2025-03-16T18:08:59.943Z [main] info: Setting log level to debug
/opt/eda-modbus-bridge/dist/app/modbus.js:357
    return device.startsWith('/') || device.startsWith('tcp://');
                  ^

TypeError: device.startsWith is not a function
    at validateDevice (/opt/eda-modbus-bridge/dist/app/modbus.js:357:19)
    at /opt/eda-modbus-bridge/dist/eda-modbus-bridge.js:97:38
    at Object.<anonymous> (/opt/eda-modbus-bridge/dist/eda-modbus-bridge.js:206:3)
    at Module._compile (node:internal/modules/cjs/loader:1356:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
    at Module.load (node:internal/modules/cjs/loader:1197:32)
    at Module._load (node:internal/modules/cjs/loader:1013:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:128:12)
    at node:internal/main/run_main_module:28:49

Node.js v18.19.0
`

So the shebang helped a bit but should I add something to my environment variables etc?

Jalle19, 2025-03-17:
You should run it using `node /opt/eda-modbus-bridge/dist/eda-modbus-bridge.js`, then you don't need to add a shebang.

What parameters did you pass to it to make it fail with `TypeError: device.startsWith is not a function`? I'm not able to reproduce that.

juntta, 2025-03-17:
> You should run it using `node /opt/eda-modbus-bridge/dist/eda-modbus-bridge.js`, then you don't need to add a shebang.

Actually I ran it with /usr/bin/node /opt/eda...

> What parameters did you pass to it to make it fail with `TypeError: device.startsWith is not a function`? I'm not able to reproduce that.

I have -d tcp://enervent_local_ip_address as parameter which triggers that if clause

I can check later tonight the actual parameters, now I cannot access the server

juntta, 2025-03-17:
> What parameters did you pass to it to make it fail with `TypeError: device.startsWith is not a function`? I'm not able to reproduce that.

Here is my terminal command I used:

`/usr/bin/node /opt/eda-modbus-bridge/dist/eda-modbus-bridge.js -d tcp://_[enervent_ip]_:502 -mqttBrokerUrl mqtt://_[broker_ip]_:1883 -mqttUserName _[usr]_ -mqttPassword _[pwd]_`

Jalle19, 2025-03-18:
I'm unable to reproduce any of this with a Debian 12 installation

```
# node dist/eda-modbus-bridge.js -d tcp://127.0.0.1:502
2025-03-18T11:02:47.576Z [main] info: Opening Modbus connection to tcp://127.0.0.1:502, slave ID 1, 5 second timeout
node:internal/process/promises:288
            triggerUncaughtException(err, true /* fromPromise */);
            ^

Error: connect ECONNREFUSED 127.0.0.1:502
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1549:16) {
  errno: -111,
  code: 'ECONNREFUSED',
  syscall: 'connect',
  address: '127.0.0.1',
  port: 502
}

Node.js v18.19.0
```

Can you try removing the `dist/` directory and then run `npm run build` again?

juntta, 2025-03-18:
Thanks @Jalle19 for your help.

I'll test removing dist and make the build again tonight and get back to you!

juntta, 2025-03-19:
Unfortunately same error when I ran same command as above after I removed the dist directory and run build again.

Jalle19, 2025-03-20:
How have you installed Node.js?

juntta, 2025-03-20:
sudo apt install nodejs -y

And npm installed same time as well

sudo apt install npm -y

Jalle19, 2025-03-20:
Would be interesting to reproduce the "startsWith is not a function", can you post the actual command you use? No placeholders etc., the real full command.

juntta, 2025-03-20:
here you go:

`sudo /usr/bin/node /opt/eda-modbus-bridge/dist/eda-modbus-bridge.js -d tcp://165.54.24.20:502 -mqttBrokerUrl mqtt://165.54.14.28:1883 -mqttUserName mosquitto -mqttPassword dressage8INERTIAL.upset#vpn -v
`

Jalle19, 2025-03-21:
Okay, that was a weird one. The problem is that you're using invalid configuration options. Instead of `-mqttBrokerUrl` you need to use `--mqttBrokerUrl` etc. The arguments parsing treats `-mqttPassword` as `-m -q -t -t -P -a -s -s -w -o -r -d`, the last one being `-d` which is the same as `--device`.

juntta, 2025-03-21:
Weird indeed and you were right, those errors went away as I added second "-". 

Now have to figure out why my mqtt broker is refusing connection...

Jalle19, 2025-03-21:
Try adding single quotes around your pasword, i.e. `--mqttPassword 'password'`. Some characters may need to be escaped with a backslash as well. Change to a simpler password without funny characters if nothing else helps.

Jalle19, 2025-03-22:
https://github.com/jestjs/jest/issues/13611 found a similar issue, will try to fix the argument parsing so it throws some kind of error.

juntta, 2025-03-22:
> Try adding single quotes around your password, i.e. `--mqttPassword 'password'`. Some characters may need to be escaped with a backslash as well. Change to a simpler password without funny characters if nothing else helps.

I already quickly tried with double quotes last night but will try single quotes today.

Password is same as before and worked with my old server so I doubt the actual password is the issue here.

juntta, 2025-03-22:
single quotes or escaping double quotes didn't work either when passed username & password as parameters. But when added crdentials directly to eda-modbus-bridge.js -file it works.

So I got it up and running again so can live with it like this but can test if you have fix in mind

closed 2025-03-22 by Jalle19