Jalle19/eda-modbus-bridge #155: Add support for Control steps of temperature and clock setup

Th1nkki, 2025-08-31:
Would it be possible to add register

> 3x0045 	Control steps of temperature 	"Here you can see the condition of temperature control. 
0=nothing used, 
1=cooling, 
2=heat recovery, 
4=heating, 
5=step delay state,
6=Summer night cooling, 
7=Startup, 8=Stop, 
9=HR clean, 
10=EXT unit defrost

This would enable monitoring and, through this, control of heating, heat recovery, and summer-night cooling.

Another register that would be useful to read and write because the clock on my device is wrong and this causes the timings to go wrong.

> 	3x0037 	Sec clock 	RTC circuit, clock seconds 	Holding register 	R 		0 - 59
	3x0038 	Min clock 	RTC circuit, clock minutes 	Holding register 	R 		0 - 59
	3x0039 	Hour clock 	RTC circuit, clock hours 	Holding register 	R 		0 - 23
	3x0040 	Day 	RTC circuit, clock date 	Holding register 	R 		1 - 31
	3x0041 	Month 	RTC circuit, clock month 	Holding register 	R 		1 - 12
	3x0042 	YEAR 	RTC circuit, clock year. Year = 2000 + register value 	Holding register 	R		
	3x0043 	Weekday 	Weekday, Mon=1, Tue=2, ... Sun=0 	Holding register 	R 		0 - 6

> 	3x0582 	C MIN RTC 	Time setting that updates RTC circuit (min) 	Holding register 	R 	W 	0 - 59
	3x0583 	C HOUR RTC 	Time setting that updates RTC circuit (tunnit) 	Holding register 	R 	W 	0 - 23
	3x0584 	C DAY RTC 	Day setting that updates RTC circuit 	Holding register 	R 	W 	1 - 31
	3x0585 	C MONTH RTC 	Month setting that updates RTC circuit 	Holding register 	R 	W 	1 - 12
	3x0586 	C YEAR RTC 	Year setting that updates RTC circuit, Year = register value + 2000 	Holding register 	R 	W	
	3x0587 	C WEEK RTC 	Weekday, Mon = 1, Tue = 2, ... Sun = 0, caluculates itself 	Holding register 	R

Jalle19, 2025-09-01:
I think one option for handling the clock would be to add an option that when enabled synchronizes the time on the ventilation unit with the time on the computer.

Th1nkki, 2025-09-01:
That's true.

Th1nkki, 2025-09-13:
> [@Th1nkki](https://github.com/Th1nkki) I've added "temperature control steps" in a branch, but at least for me it seems to always have a value of 0. Is that your experience as well?

"When I use Talologger software, I can read the values just fine. I can read heat recovery and heating, but if I understand correctly, you get it to work?"

Jalle19, 2025-10-04:
Been trying this out now for some time, just merged it.

[image]